lkesteloot/alice #8: Implement pattern

bradgrantham, 2017-04-30:
- [ ] Implement pattern in rasterizer
- [ ] Couple protocol to hardware (protocol may be changed) - protocol.cpp test?
- [ ] Integrate command into software
  - [ ] hardware_rasterizer.cpp
  - [ ] parse_gpu_protocol

lkesteloot, 2017-05-01:
It's hard to tell if pattern is working because of my double-pixel issue. I'll fix that next. Not sure what you mean by "couple protocol to hardware", but the idea of a protocol.cpp test is good. I'll do that after I fix the double-pixel so that we can actually see the real pattern. By the way, I define "y = 0" as being the top line on the screen for pattern lookups. So the first pattern in the array in C would be on the top line of the screen. I can invert that if you'd like, though it likely doesn't matter at all for any demo we have.

bradgrantham, 2017-05-01:
I think I meant that you said you were processing the pattern command but
not doing anything with it.  Maybe I should have said "do something with
the pattern the FPGA processed."

The pattern is defined bottom-to-top
<http://inetsd01.boulder.ibm.com/pseries/en_US/infocenter/base/43_docs/libs/gl32tref/defpattern.htm>,
but I wouldn't fix it.  No one will notice.  If we do decide it's important
for some reason, we can invert it hardware_rasterizer.c; 16 divides evenly
into 480.

If we get really ambitious and try to port more demos that use non-16x16
patterns, we can worry about it then.

    -Brad

lkesteloot, 2017-05-01:
Can I close this issue then? I saw that you made changes to the parser but I didn't test them.

bradgrantham, 2017-05-01:
Sure.  If the parser doesn't work, I'll open it as a bug.

closed 2017-05-01 by lkesteloot