lkesteloot/alice #13: Implement Z-buffer

lkesteloot, 2017-05-01:
This is huge, because naively I'd do a Z-buffer pixel read each time I needed it, but the latency there is huge. Maybe initially do it that way. But then I either need to pre-fetch each row of the bounding box, which is wasteful since at least half of those (on average) won't be necessary, or do something even more complex. And remember that we also have to write the updated Z values back. Perhaps add another port to the memory controller so that we can do that simultaneously with writing color.

lkesteloot, 2017-05-01:
The Z buffer code in fpgasim does a divide per pixel. So does the version in my graphics engine. Is that really necessary? Did they always do that, like back in 1985? @bradgrantham

bradgrantham, 2017-05-01:
So, are we talking about DisplayImage.m:261..263?  I think you could divide your w variables by the triangle area in triangle setup, since area is a constant per triangle.

Here's some more I wrote before looking at your code, which I now think wasn't what you meant:

I don't think the raster engines in 1985 did a divide per pixel!  The first SGI to support perspective correct texturing was the [VGXT](http://www.sgistuff.net/hardware/graphics/powervision.html), and that was produced after the VGX in 1990.

I think the secret is that depth-buffering and Gouraud interpolation aren't affected much by perspective.  I believe that's because without texture-mapping there isn't much detail inside a triangle and your eye just doesn't detect wrong interpolation like it will with a texture map.

I think engines through the 1990's did, or at least approximated it.  All hardware these days does perspective correct interpolation for the triangle interpolants, and thus all attributes are perspective correct.

I would call perspective correction an optional feature.  To do it really right, to my understanding, libgl would need to pass the rasterizer W; X/W, Y/W, Z/W, and 1/W would all be interpolated, and then you'd divide by interpolated 1/W.  Or something.  I'm going partly off [this scratchapixel.com blog entry](https://www.scratchapixel.com/lessons/3d-basic-rendering/rasterization-practical-implementation/perspective-correct-interpolation-vertex-attributes).

bradgrantham, 2017-05-01:
An idea about doing the Z-buffer with long fetch times would be to maybe implement FIFOs between a rasterizer block and a depth-comparison block.  If your FIFO is big enough, maybe you can hide that latency.  You could also have a FIFO for writing the results, if the depth test was disabled or passed, to hide write latency.  Wait, don't you already have a write FIFO?

lkesteloot, 2017-05-01:
You're right about the divide per pixel. So odd that I saw the divide per pixel for color and thought, "Oh, that's constant for the triangle", but then saw the exact same divide for Z and didn't make the same connection. So that's solved. I just need to compute the reciprocal of area once per pixel.

I don't have a write FIFO. I write as I go. I could have a write FIFO, and make sure to flush it before moving on to the next triangle (in case they overlap).

commit 8e61fb83 (2017-05-26): Fix Z buffer artifacts. Fixes #13.
closed 2017-05-26 by lkesteloot