lkesteloot/alice #19: Have U-Boot or EPCS load the bitfile on boot

bradgrantham, 2017-05-03:
Need to recompile uboot so that fpga2sdram_handoff=0x00000333 or put that by hand in the bootcmd variable

lkesteloot, 2017-06-02:
To do:

- [ ] Change bridge to 0xFFF.
- [ ] Convert bitfile to correct format.
- [ ] Put bitfile in file system.
- [ ] Load bitfile in U-Boot.
- [ ] Fix dip switches on board.
- [ ] Load bitfile into FPGA.

lkesteloot, 2017-06-02:
According to [this page](https://rocketboards.org/foswiki/Documentation/LoadingFPGAFromUBoot), we want an RBF file:

    quartus_cpf -c <input file>.sof <output file>.rbf

Switches:

    up-up-down-up-up-up = uncompressed rbf

lkesteloot, 2017-06-02:
Also: [Programming FPGA from HPS](https://rocketboards.org/foswiki/view/Documentation/GSRD131ProgrammingFPGA).

lkesteloot, 2017-06-03:
Change 296bfd27 adds the boot script. The Google doc has the rest of the info.

closed 2017-06-03 by lkesteloot