lkesteloot/alice #32: Cut TFT friend 50mA bridge and one 25mA bridge

bradgrantham, 2017-05-04:
This should bring us comfortably under the PowerBoost 1000C charge limit and allow us to run on battery if desired.

lkesteloot, 2017-05-04:
Is this better than using PWM to dim it?

bradgrantham, 2017-05-04:
If the device could be rigged to immediately come up in a lower-power mode,
that would be fine, I think.  Otherwise it comes up in a mode exceeding the
battery board's capacity.  Could we weakly pull-down the backlight enable
so the backlight comes up completely off, and could the FPGA bitfile
initialize the LCD backlight PWM at 40%?

It's true that just using PWM would mean we could drive it at 100% of 125mA
if we wanted to.  The DE0 would have to be plugged into the wall wart
then.  I think if we wire the DE0 to the battery board, then it can't draw
more than 5W even if connected to USB power.

This is all assuming 3W of power draw is just the backlight.  I guess I can
set brightness to 100% and 0% now and see what the power draw is, if I can
get a reliable, robust current or power measurement

Am I putting too much work into this?  I kind of feel like we're close to a
power envelope that could run from battery, so I'm more encouraged to try.
It's not a priority, but I hope by the Geek Retreat I can get to trying all
the techniques to lower power and then try running from battery.  I realize
tilt/gyro is a much higher priority, and I plan to look at that over the
weekend.

lkesteloot, 2017-05-04:
> Could we weakly pull-down the backlight enable

I don't think so, it's PWM. But we could perhaps pull-down the "display on" pin, so that the display is completely off until we're ready to run its PWM. It's a bit risky, though! What if we use PWM and re-load the FPGA right when the PWM is on "high"? Then for a few seconds the LCD will pull full current.

Probably best to do as you say and remove the bridges.

bradgrantham, 2017-05-28:
Display is not very bright with only one 25mA bridge closed, but it's probably bright enough.

The display is easier to see from the top, so I'm kind of thinking that we may want to rotate the LCD.  See #41.

closed 2017-05-28 by bradgrantham