lkesteloot/alice #33: Split out GPU functions into a separate library

bradgrantham, 2017-05-04:
Right now there's the potential for drift between the gpu code in hardware_rasterizer.c and the other files.

All code interacting with the GPU should probably be in one C file.

lkesteloot, 2017-05-04:
You mean the `cmd_draw()` functions, etc? I can take this one.

bradgrantham, 2017-05-04:
Yeah and some sloppy gpu_ functions I wrote to do think like start the gpu,
wait on F2H_BUSY, etc.  And it would be nice to wrap access to gpo and gpi
to keep track of the previous brightness, etc.  Thank you!

<https://github.com/lkesteloot/alice/issues/33#issuecomment-299283876>, or mute
the thread

bradgrantham, 2017-05-05:
Mind if I do this over the weekend while you're gone?

lkesteloot, 2017-05-05:
Fine by me.

commit 029d9379 (2017-06-04): Use awesome library from libgl. Fixes #33.
closed 2017-06-04 by lkesteloot