lkesteloot/alice #68: Reference rasterizer rectf*() GL API not functional in flight demo

sgi-demos, 2023-03-20:
Hello!

In debugging the flight demo for [sgi-demos](https://github.com/sgi-demos/sgi-demos) project, I found the runways and taxiways were not drawing due to a bug in rectf_() in [gl.c](https://github.com/lkesteloot/alice/blob/master/alice4/software/libgl/gl.c):

```
void rectf_(Coord x1, Coord y1, Coord x2, Coord y2) {
    pmv_(x1, y1, 0);
    pdr_(x1, y2, 0);
    pdr_(x2, y2, 0);
    pdr_(x2, y1, 0);
    end_polygon();
}
```

And this fixed it for me:

```
void rectf_(Coord x1, Coord y1, Coord x2, Coord y2) {
    bgnpolygon();
    pdr_(x1, y1, 0);
    pdr_(x1, y2, 0);
    pdr_(x2, y2, 0);
    pdr_(x2, y1, 0);
    endpolygon();
}
```

Thanks for the inspiration and the GL rasterizer!

lkesteloot, 2023-03-20:
Nice! Yeah the flight code is clearly half-way through some larger change, unfortunately! It's a snapshot we found on some FTP site. I'm a little surprised we've not found a better version.

bradgrantham, 2023-03-20:
That's a great catch!  Thanks, fixed and pushed.

I wonder if "pclos" also has a bug.  "jello" calls "pclos" but I'm not sure
when I can find time to test it.

    -Brad
--
Brad Grantham, http://plunk.org/~grantham/

sgi-demos, 2023-03-20:
> I'm a little surprised we've not found a better version.

I've got four versions of the source here, and I intend to port _all_ of them. :^)

> I wonder if "pclos" also has a bug.

Oh good point, that may help with another glitch I'm seeing in newave.  Based on the GL reference, looks I could funnel pmv*() to bgnpolygon() + v3f() and pclos() to endpolygon().  

Do you recall any gotchas with doing this?  I realize it may have been a few years.